eirikpre/VSCode-SystemVerilog #167: Extension seems to slow down fuzzy searching

sunnydxp, 2022-05-27:
Issue Type: <b>Performance Issue</b>

After using extension bisect, I found that this extension is slowing down vscode's Ctrl+P. Not sure what other information I can provide at this time. Let me know.

Extension version: 0.13.2
VS Code version: Code 1.67.2 (c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5, 2022-05-17T18:20:57.384Z)
OS version: Darwin x64 21.4.0
Restricted Mode: No
Connection to 'ssh-remote+180' could not be established
Remote OS version: Linux x64 4.18.0-348.20.1.el8_5.x86_64

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz (12 x 2600)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: disabled_off<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>skia_renderer: enabled_on<br>video_decode: enabled<br>video_encode: enabled<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|4, 4, 3|
|Memory (System)|16.00GB (0.83GB free)|
|Process Argv|--crash-reporter-id f619db14-a818-410d-b452-772c1ad0d7de|
|Screen Reader|no|
|VM|0%|

Connection to 'ssh-remote+180' could not be established

|Item|Value|
|---|---|
|Remote|SSH: 180|
|OS|Linux x64 4.18.0-348.20.1.el8_5.x86_64|
|CPUs|Intel(R) Xeon(R) Gold 6242 CPU @ 2.80GHz (128 x 3900)|
|Memory (System)|1510.06GB (1141.16GB free)|
|VM|0%|
</details><details>
<summary>Process Info</summary>

```
CPU %	Mem MB	   PID	Process
   63	  1393	 79654	code main
    6	    82	 79658	   gpu-process
    0	    33	 79659	   utility-network-service
   22	  1655	 79661	   window (xvss_ip_ws1 (Workspace))
    1	   115	 79670	   shared-process
    0	    16	 79672	     ptyHost
Connection to 'ssh-remote+180' could not be established

Remote: SSH: 180
CPU %	Mem MB	   PID	Process
    0	     0	3136344	remote agent
    0	     0	3136452	   ptyHost
    0	     0	3323067	     /bin/bash
    0	     0	3323275	     /bin/bash
    0	     0	3322896	   extensionHost
    0	     0	3323454	     /xpeng/asic/users/arund/.vscode-server/bin/c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5/node /xpeng/asic/users/arund/.vscode-server/extensions/eirikpre.systemverilog-0.13.2/dist/server/server.js --node-ipc --clientProcessId=3322896
    0	     0	3332518	     /xpeng/asic/users/arund/.vscode-server/bin/c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5/node /xpeng/asic/users/arund/.vscode-server/bin/c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5/extensions/json-language-features/server/dist/node/jsonServerMain --node-ipc --clientProcessId=3322896
    0	     0	3322955	   fileWatcher
    0	    16	3344844	   extensionHost
    0	     0	3345132	     /xpeng/asic/users/arund/.vscode-server/bin/c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5/node /xpeng/asic/users/arund/.vscode-server/extensions/eirikpre.systemverilog-0.13.2/dist/server/server.js --node-ipc --clientProcessId=3344844
    0	     0	3353661	   /usr/bin/ps -ax -o pid=,ppid=,pcpu=,pmem=,command=
```

</details>
<details>
<summary>Workspace Info</summary>

```

Connection to 'ssh-remote+180' could not be established
|  Remote: SSH: 180|    Folder (xvss_ip_ws2): 2235 files|      File types: tcl(273) db(247) sv(189) sdb(154) v(150) so(70) json(53)
|                  log(35) svh(32) upf(28)
|      Conf files: makefile(4);
```

</details>
<details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383cf:30185419
vspor879:30202332
vspor708:30202333
vspor363:30204092
vstes627:30244334
pythonvspyl392:30443607
pythontb:30283811
pythonptprofiler:30281270
vshan820:30294714
vstes263:30335439
vscoreces:30445986
pythondataviewer:30285071
vscod805cf:30301675
pythonvspyt200:30340761
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
vsaa593cf:30376535
vsc1dst:30438360
pythonvs932:30410667
wslgetstarted:30449410
pythonvsnew555:30457759
vscscmwlcmt:30465135
cppdebug:30492333
vsclangdc:30486549

```

</details>

joecrop, 2022-05-27:
Seems similar to #165 with regard to the remote SSH issue.

joecrop, 2024-08-30:
I believe this is fixed after version 0.13.3

closed 2024-08-30 by joecrop