eirikpre/VSCode-SystemVerilog #188: Conditional assignments within a port instantiation break syntax highlighting pattern on the next line

birdybro, 2022-10-26:
https://github.com/MiSTer-devel/SMS_MiSTer/blob/master/SMS.sv#L477-L478

[image]

joecrop, 2023-01-22:
I'll be honest, this is low on the priority list because must commercial linting tools would scream bloody murder when you put this amount of logic directly in a port instantiation.  An easy fix is to just assign this into a net before passing it to the port.

birdybro, 2023-01-23:
Agreed on principle for sure, but I figured I'd report it anyways regardless as it was unexpected behavior.

[This extension](https://github.com/mshr-h/vscode-verilog-hdl-support) does handle it correctly, but I prefer yours :P

joecrop, 2023-09-12:
OK, I finally had time to dig into this and I have some bad... well maybe good? news. I can't recreate the problem in a VSCode environment with just the systemVerilog extension installed.

[image]

I wonder if you have another extension installed that is either overriding this extension or conflicting wtih it.

birdybro, 2023-09-13:
I just disabled the other extension from Masahiro Hiramori, I reinstalled yours, and the issue comes back. Maybe your color scheme youa re using doesn't differentiate between the two? Also is this the current release that is up on the vscode extension store you are using? I'm using the Dracula Pro color scheme which might make a distinction between two elements and yours might treat them as the same color... hrmmm

[image]

I used the file from the time I made this issue --> https://github.com/MiSTer-devel/SMS_MiSTer/blob/901ff4b5683d12929614c4250d13c4ca4e8b62b0/SMS.sv

Using the latest version makes no difference, just wanted to make sure there wasn't a change in code that somehow altered this.

Just confirmed, I changed to Dark Modern (not part of the Dracula color scheme) and sure enough they are the same color now. So whatever special distinction dracula makes between these two is not represented in probably a lot of other color schemes. Not a false positive, but probably not a thing most people would see unless they happened to use your extension combined with using Dracula color themes.

I don't know how to debug linters to see how the output differs to the perspective of how the color scheme is applied to different elements, or else I would help provide more info.

https://draculatheme.com/visual-studio-code - Here is the theme to recreate the issue.