eirikpre/VSCode-SystemVerilog #191: Auto Indexing continues to work after disableIndexing

Karl-Han, 2023-03-02:
Platform: macOS 13.0.1 (22A400)
VSCode Version: 1.75.1
Plugin Version: v0.13.3

Here is my configuration related to SystemVerilog Extension: 
```
    "systemverilog.formatCommand": "verible-verilog-format --indentation_spaces 4",
    "systemverilog.disableIndexing": true,
    "systemverilog.compileOnSave": false,
```

With customized formatCommand, I can confirm that this configuration is loaded by SystemVerilog, but every time I switch to another file or other panel and switch back, SystemVerilog will automatically indexing the current file, which is not what I expected with `systemverilog.disableIndexing`. 

I want to disable indexing and use Control Palette to index by demand. Is this a bug or my configuration is wrong?

joecrop, 2023-03-03:
If memory serves me correctly. The `disableIndexing` setting will prevent the workspace from being crawled for sv files that you may not have opened yet. Files are always indexed on open.

Is there a command that exists today that let's you manually trigger the indexing, or are you requesting that feature too?

Karl-Han, 2023-03-03:
Actually, from your expression, I think I got the meaning of `disableIndexing` wrong. 
> systemverilog.disableIndexing: Boolean, Disable indexing

I thought it means disable global indexing by looking into README. If it is possible, I like to request for that global indexing disable feature :), since it is friendly to computation source.

joecrop, 2023-03-19:
Hi @Karl-Han, After looking into implementing this, I think there is already a setting for it. Will the `systemverilog.enableIncrementalIndexing` setting do what you are hoping?

```js
"systemverilog.enableIncrementalIndexing": {
    "type": "boolean",
    "default": true,
    "description": "Enable incremental indexation as you open files."
},
```

Karl-Han, 2023-09-07:
After configuring this option, it works as expected without automatic indexing.

closed 2023-09-07 by Karl-Han