eirikpre/VSCode-SystemVerilog #204: Does the 'systemverilog.excludeIndexing' work well?

diverger, 2023-08-19:
Hi,
I try to add a foler to the excluding list as "folder_name/**', but it seems don't work. The output window of the vscode still shows the indexing informations. My platform is Windows.

Thanks.

diverger, 2023-08-23:
@joecrop Hi,

1. Yes I closed and reopened the vscode after changing the settings.
2. The folder indeed in the root of the workspace.
3. The Vivado compiling well if I close the vscode. But I don't tried to disable this single plugin. I will try it latter.

The reason I think it's caused by this plugin, is when do Vivado sythesis, the systemverilog plug window still refresh indexing informations  continuously, such as:

SystemVerilog: Indexing: Unable to process file: blablabla....

And the files under the "folder_name" which I disabled the indexing are shown there however. Though the status bar showed the plugin only indexed 12 objects.

And I'm a bit confused now. As the status bar showed there are only '12' objects indexed (12 is reasonable, because these are the sources I need the plugin to parser). But why the window still show indexing informations about other files? It seems, every time Vivado touch a file, the vscode or the plugin take a look at it, and then it may index it or abandon it. Just a guess. :)

Thanks.

joecrop, 2023-08-23:
OK, In that case, I think adding a `**/` in front of the path will help. Looking at the code for this, I think the paths it is comparing with are actually the full paths on the disk, not the path relative to your workspace root.

diverger, 2023-08-24:
I tried it. It didn't work. When Vivado doing Synthesis, the SytemVerilog output windows showing the indexing infos continuously as before. And finally show:

_[Designutils 20-411] The directory 'blablabla/realtime' could not be deleted and may be locked. This directory may need to be manually deleted to continue._

The same result with my previous settings.

joecrop, 2023-08-28:
OK, new Idea!

I wonder if it is a Windows path issue. Can you try a couple permutations?

1. `**\folder_name\**`
2. `**\\folder_name\\**

If either of those don't work, I think the code will have to be modified to handle the paths in an OS-specific way.

joecrop, 2023-09-12:
I just played around with this in Windows (finally had time). I was able to get the extension to stop indexing when I set: 
 **Exclude Indexing** = `**/folder_name/**`

closed 2024-01-05 by joecrop