eirikpre/VSCode-SystemVerilog #206: SystemVerilog: Indexing......

the-moog, 2023-09-29:
This appears to be a blocking thread. On a large filesystem where  symlinks in the tree result in many files being indexed (about 10k) results in it hanging.  Make this a background task or time limit it, clipping at, e.g. max depth 5 seconds?

Pressing cancel each time does stop the process.

souranild, 2023-10-19:
Same issue found. Please let me know if there is a workaround

joecrop, 2024-01-05:
If you have a really large workspace, you can change the extension settings I pasted below. If you have a huge workspace and WANT to parse every file with full precision, then it will be slow. If you only want to parse files when you open them, it will be fast. There are many flavors in between. The most powerful tool is to use `systemverilog.includeIndexing` and `systemverilog.excludeIndexing`. Feel free to re-open this ticket if this explanation is insufficient.

- `systemverilog.includeIndexing`: _Array_, Files included for indexing (glob pattern). Examples:
  - Include files within the workspace's rtl folder (`*` at front of pattern denotes path is relative to workspace root): `**/rtl/**/*.{sv,v,svh,vh}`
  - Add all files with a '.svp' extension: `**/*.svp`
  - Add files only when in a specific workspace: `/abs/path/to/workspace/rtl/**/*.{sv,v,svh,vh}`
- `systemverilog.disableIndexing`: _Boolean_, Disable indexing
- `systemverilog.excludeIndexing`: _String_, Exclude files from indexing based on glob
- `systemverilog.forceFastIndexing`: _Boolean_, Use fast regular expression parsing
- `systemverilog.enableIncrementalIndexing`: _Boolean_, Enable incremental indexation as files are opened
- `systemverilog.parallelProcessing`: _Integer_, Number of files to process in parallel during indexing
- `systemverilog.forceFastIndexing`: _Boolean_, force indexer to bo basic parsing. Good for when the extension takes too long to initialize.
- `systemverilog.enableIncrementalIndexing`: _Boolean_, Enable incremental indexation as you open files.
- `systemverilog.maxLineCountIndexing`: _Boolean_, When indexing a file, if the line count is larger than this number, _fast_ indexing will be used to improve symbol lookup performance, as fewer symbols will be parsed.
- `systemverilog.documentSymbolsPrecision`: _String_, The level of detail the parser should use when looking for symbols:
  - _full_: detect blocks, ports, parameters, classes, methods, typedefs, defines, labels, instantiations, assertions, and _references across files_.
  - _full_no_references_: detect blocks, ports, parameters, classes, methods, typedefs, defines, labels, instantiations, and assertions.
  - _declarations_: detect blocks, ports, parameters, classes, methods, typedefs, and defines.
  - _fast_: detect only common blocks (module, class, interface, package, program) without hierarchy.

closed 2024-01-05 by joecrop