eirikpre/VSCode-SystemVerilog #225: Autocopletion does not work

BLu85, 2024-04-09:
Hi,

When this extension is not installed, word suggestions are shown while typing (which is what I want).
When this extension is installed, the drop down suggestion menu stops working and no words are suggested.

Is there a clash between this extension features and the IntelliSense behaviour in Vscode?
Do I have to set some setting in order to have it working?

Thanks.

joecrop, 2024-05-28:
This extension definitely supports autocompletion, and in fact enhances it with more tokens.

Can you please try using this extension in a vscode environment with no other extensions installed? Perhaps you have an extension installed that is conflicting with this extension? The other thing to try is the [recommendations](https://github.com/eirikpre/VSCode-SystemVerilog?tab=readme-ov-file#recommendations).

BLu85, 2024-05-29:
Thanks for helping.
Reading the recommendations I found out that in my `settings.json` I had
```
    "[systemverilog]": {
        "editor.tabSize": 2,
        "editor.quickSuggestions": {
          "comments": "off",
          "strings": "off"
          "other": "off"
        },
        "editor.formatOnType": true
    },
```

I have now set to `off` only `others` and the extension is suggesting words as expected.

Thanks

closed 2024-05-29 by BLu85