eirikpre/VSCode-SystemVerilog #231: Verilator problem matcher regex doesn't actually detect anything

alex-the-new-guy, 2024-07-18:
Windows 11, using latest version of Verilator and extension, both build from git today. 

Test modules are:
top.v
```
module top(
  input wire test1,
  input wire [31:0] test2
);

submodule my_submodule(
  .test_input(1'b0),
  .test_output(test2),
  .non_existing_input(1'b0)
);

endmodule
```

submodule.v
```
module submodule
(
  input wire [23:0] test_input, // test input comment

  output reg [7:0] test_output // test output comment
);

endmodule
```

Verilator output is:

```
%Error-PINNOTFOUND: C:\tmp\vscode_verilog_testenvs\verilator/sources\top.v:9:4: Pin not found: 'non_existing_input'
    9 |   .non_existing_input(1'b0)
      |    ^~~~~~~~~~~~~~~~~~
                    ... For error description see https://verilator.org/warn/PINNOTFOUND?v=0.000
%Error-ASSIGNIN: C:\tmp\vscode_verilog_testenvs\verilator/sources\top.v:8:16: Assigning to input/const variable: 'test2'
    8 |   .test_output(test2),
      |                ^~~~~
%Error: Exiting due to 2 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Which leads to no problems being detected.
From what I've deduced from using a regex debugger, the issue comes from file line being included with column name and general mismatch between match groups in regex and sections

Adding regex to match Verilog and SystemVerilog file extensions so that problem matcher objects are as listed below, seems to fix the issue.

package.json
```
            {
                "name": "verilator-error",
                "owner": "Verilator",
                "source": "systemverilog",
                "severity": "error",
                "fileLocation": "autoDetect",
                "pattern": [
                    {
                        "regexp": "%Error(-[A-Z0-9]+)?: ((\\S+((\\.sv)|(\\.v))):(\\d+):((\\d+):)? )?(.*)$",
                        "severity": 1,
                        "file": 3,
                        "line": 7,
                        "column": 9,
                        "message": 10
                    }
                ]
            },
            {
                "name": "verilator-warning",
                "owner": "Verilator",
                "source": "systemverilog",
                "severity": "warning",
                "fileLocation": "autoDetect",
                "pattern": [
                    {
                        "regexp": "%Warning(-[A-Z0-9]+)?: ((\\S+((\\.sv)|(\\.v))):(\\d+):((\\d+):)? )?(.*)$",
                        "severity": 1,
                        "file": 3,
                        "line": 7,
                        "column": 9,
                        "message": 10
                    }
                ]
            },
```

alex-the-new-guy, 2024-08-30:
> @alex-the-new-guy will you please test the update I made? I have compiled an extension with the fix here: #237 To maintain backward compatibility with older Verilator version, I kept the old regexes and added yours to the list.

So, I've been doing a bit of a refactoring of the parser pipeline. Veeery slowly bc busy with work and university. I hope to finish it over the weekend and I can incorporate the changes you've made if you give me link to repo.

joecrop, 2024-08-30:
@alex-the-new-guy the proper merge procedure is:
1. Create a fork of the repo that is owned by you: https://github.com/eirikpre/VSCode-SystemVerilog/fork
2. Create a new branch with your changes and push it up to YOUR repo
3. Create a pull request in this repo that will merge your branch into the master branch of this repo: https://github.com/eirikpre/VSCode-SystemVerilog/compare

I'll leave my changes in master for now, and you can modify them as part of your pull request. I only changed one file: https://github.com/eirikpre/VSCode-SystemVerilog/pull/237/files#diff-7ae45ad102eab3b6d7e7896acd08c427a9b25b346470d7bc6507b6481575d519