eirikpre/VSCode-SystemVerilog #233: Syntax highlighting in comments

arquer, 2024-08-05:
keywords seem to be highlighted even in comments
[image]

joecrop, 2024-08-30:
Are you sure this isn't coming from a different extension? I don't see that behavior when I run a clean vscode with only this extension.
[image]

arquer, 2024-08-30:
hmm I'll do some more checking

Muione, 2024-10-11:
I have met the same issue that keywords  is highlighted  in comments, for eaxmple:
[image]

arquer, 2024-10-11:
I have noticed that this often happens when you have just opened a file and the parser is still figuring it out. If you let it work for a bit (and specially if you scroll up and down these tend to go away).

I am closing this.

closed 2024-10-11 by arquer