eirikpre/VSCode-SystemVerilog #245: SysteVerilog: 0 indexed objects (Urgent)

jbengineer, 2024-12-28:
with recent changes /updates, It seems like object indexing stopped working. I trided to go back with previous version of VSCODE and even this extension. nothing works.

joecrop, 2025-01-06:
The latest extension and latest version of VSCode are working for me. Perhaps you can try resetting the extension settings to their default? Even if you re-install the extension, the settings will not reset. You need to go into the settings and manually reset them.

jbengineer, 2025-01-09:
Thanks @joecrop for your response. I tried various suggestions you purposed but there is no luck. With further debug I found that this issue is only present with SAMBA mapped network drive folders. Same folder copied to to C:\xx folder indexes the objects without any issue. Please let me know if anybody seen this issue.

jbengineer, 2025-01-31:
I had my code base in a Linux machine with Samba mapping to windows 11. To fix this issue the read\write ="true" privilege must be set in Linux for this whole folder.
I think this showed up due to some window 11 update which does not allow Samba to work the way it used to work